ZhuangLab/storm-control #1: a hardware list that HAL4000 current drives

coronin, 2015-01-30:
I would really like to have HAL4000 to drive our STORM imaging. From what I heard, it is much easier to use the exact hardware that your lab currently is using. Is it possible to add a hardware list that HAL4000 current drives? Or, a published protocol? Thanks!

HazenBabcock, 2015-01-30:
You can see what hardware we are using by browsing the hal4000/xml/xxx_hardware.xml files. The <module_name> tag tells you what class is used. In the corresponding file for that class you can see what hardware is used. 

All of the hardware control classes are in the sc_hardware folder ordered by manufacturer name. If you don't see your hardware there it is probably not supported.

Feel free to e-mail me with a list of the hardware that you have and I can try and give you some idea of how much difficulty you are likely to face.

coronin, 2015-01-30:
storm4_hardware.xml
    
    camera
    * andorCameraControl => driver from Andor Solis
    
    bluetooth
    * halBluetooth
    
    focuslock
    * focuslock.storm4FocusLockZ
    ** sc_hardware.madCityLabs.mclController => NanoDrive 2359,2636,2637
    ** sc_hardware.thorlabs.uc480Camera
    ** sc_hardware.thorlabs.LDC210
        
    illumination
    * illumination.illuminationControl
    ** storm4_illumination_settings.xml
    *** sc_hardware.thorlabs.FW102C => HalFW102C
    *** sc_hardware.crystalTechnologies.hAotf => CrystalTechAOTF64Bit
    *** sc_hardware.coherent.hLaser => CoherentCube 405
    *** sc_hardware.nationalInstruments.hNicontrol => PCI-6733 (DAC needs a connector)
    
    joystick
    * joystick.gamepad310JoystickControl
    ** sc_hardware.logitech.gamepad310
    
    misc_control
    * miscControl.storm4MiscControl
    ** sc_hardware.olympus.ix2ucb
    
    stage
    * stagecontrol.storm4StageControl
    ** sc_hardware.marzhauser.marzhauser
    
    tcpip
    * halTcpControl
    ** sc_library.tcpServer

I checked the storm4 and have a summary above. I have couple questions:

1. Our camera is DU897 Ultra, driven by Solis. For it to work, are there any parameters need to be tweaked in andorCameraControl.py ?

2. Currently, we do not have a focus lock setup for the TIRF. I would like to make the exactly one you are using. What do I need? A USB CMOS camera and a IR laser driven by LDC210 from thorlabs; a MCL NanoDrive (controller + Nano-F). Or, will the MCL C-Focus system work?

3. Our lasers are all from coherent, controlled by a home-made Arduino. Do we have to use AOTF, or a MBP 647 laser?

4. We do not need to worry about the bluetooth, stage and gamepad at current stage, correct?

HazenBabcock, 2015-01-30:
1. You should be able to use that camera without adjusting anything.

2. We don't have any experience with MCL C-Focus. It would probably be difficult to integrate as HAL has it's own focus lock event loop and you'd have to completely re-write this. Currently we are using one of two different focus-lock flavors. The simpler one uses a QPD, an IR laser driven by a LDC210 and a MCL objective positioner (see the STORM1 setup). The more complicated one uses the UC480 USB CMOS camera from Thorlabs, an IR laser driven by a LDC210 and a MCL objective positioner. Additionally we found that for the best performance it was necessary to turn the IR laser on and off at a fairly high frequency (see thorlabs.LD210.LDC210PWMxx) to destroy the laser coherence on the time scale of the camera imaging time. Describing the optical path in a comment is probably beyond me, but if you e-mail me directly I can provide you with a diagram.

3. The arduino controls the laser shutters? Or the laser powers? Both? You'll need to write a an interface class for the Arduino that HAL can call to control the illumination. This class would probably be something like sc_hardware.nationalInstruments.hNicontrol and would be a sub-class of sc_hardware.baseClasses.IlluminationHarware.

4. No you don't need to worry about these. The only thing that is required is the camera, you can remove all the \<modules\> in the hardware.xml file and the software should still work.

HazenBabcock, 2015-02-07:
I added a table in the README.md file in the sc_hardware directory that lists the current status of the various hardware control classes. Hopefully this will be helpful.

closed 2015-06-26 by HazenBabcock