ZhuangLab/storm-control #43: kilroy -  cannot be opened in test mode

BogdanBintu, 2016-11-16:
In **kilroy.py** line 42:
        if not hasattr(parameters, "num_simulated_valves"):

hasattr(parameters, "num_simulated_valves") will always return false even if 
parameters.get("num_simulated_valves") will not error and return the value in the xml loaded by the paramaters.

Suggested change:
Replace:
        if not hasattr(parameters, "num_simulated_valves"):
            self.num_simulated_valves = 0
        else:
            self.num_simulated_valves = parameters.get("num_simulated_valves")
with:
self.num_simulated_valves = parameters.get("num_simulated_valves", 0)

To get variables with defaults use: self.<variable> = get(<variable>,<default_value>)

HazenBabcock, 2016-11-17:
Please use the following procedure for your suggestions:
(1) Create a fork of storm-control (which you may have already done) on Github.
(2) Create a branch off of an updated master branch, "git checkout -b fix_kilroy_test_mode".
(3) Make the changes (and only the changes) you want to see in master.
(4) Test your changes.
(5) Create a pull request on Github for your branch.

Also you can use markdown in your issues, which makes them more legible.

```
if not hasattr(parameters, "num_simulated_valves"):
    self.num_simulated_valves = 0
else:
   self.num_simulated_valves = parameters.get("num_simulated_valves")
```

HazenBabcock, 2017-01-26:
FWIW, this actually fails because the parameters object has changed. The correct way to test for the presence of a parameter in a parameter object is:

```
parameters.has("some_attribute_name")
```

HazenBabcock, 2018-01-23:
Fixed? Closing for now.

closed 2018-01-23 by HazenBabcock