ZhuangLab/storm-control #45: rolling shutter cameras

HazenBabcock, 2016-12-01:
Need an option to allow exposure only during the time when the camera is not actually reading anything out.

BogdanBintu, 2019-04-25:
I implemented a hardware fix for this on the lumencor system. One can connect the global shutter from the BNC breakout to the camera timing. Given the appropriate polarity of the camera trigger (which can be changed) this will ensure that the lasers are triggered off for the 33ms offset of the Hamamatzu camera.
This could also work for the AOTF. Do you know if the AOTF has a global shutter input?

HazenBabcock, 2019-04-26:
I think that depends on the AOTF RF driver. The one that I have, a 97-03926-12 from Gooch and Housego has a blank pin that should serve for this purpose.

BogdanBintu, 2019-04-27:
Thanks for the help.
I have a very similar AOTF RF driver. The blank pin is pin 31. 
[image]

Suppose I connect pin 31 and presumably pin 22 (GND?) to a DO on the NI DAQ.
Then I can trigger on/off the AOTF?

BogdanBintu, 2019-04-27:
Thanks, tried it out and it works. The camera now triggers globally the AOTF.
It indeed makes all the pixels equivalent but it reduces ~33ms of the exposure time.
There might be better ways?

HazenBabcock, 2019-04-27:
Maybe? If the maximum frame rate of your camera is 100Hz full chip then the readout time would be 10ms. You'd lose that at both ends of the exposure, so I'd expect approximately 20ms reduction in exposure time.