ZhuangLab/storm-control #59: Improved tile generation in Steve

jeffmoffitt, 2017-07-21:
It would be great if the user could generate a set of tiled positions around different regions of a sample in Steve. Specifically, the user should have control over the number of positions in X and Y, as well as the spacing of these positions. With this control the user could then dynamically adjust the tile to fit different features of the sample.  

In parallel, it would be useful if Steve would display these tiles as they are being generated so that the user can confirm that they properly cover the desired features.

leonardosepulveda, 2018-04-03:
I concur. Many of my tissue samples have variable shapes, so using a rectangular grid can actually be very inefficient, many FOVs imaged would be empty. I Thought that there were two ways around this: have a simple code that blurs the image composite taken at 10x and then calculates a "mask" using a threshold, so we keep only FOVs inside that mask. The problem with this approach is that the segmentation may vary in quality. Maybe the most efficient way to achieve this is to allow for a tool that allows to draw a polygon over the image, this polygon can be used to calculate a set of tiled positions with defined spacing.

HazenBabcock, 2018-10-23:
FWIW I'm working on Steve now. Hopefully once I have Steve cleaned up and better organized it will be easier to add this feature.

HazenBabcock, 2019-10-22:
I just merged 'steve_v2' into master as I'm pretty sure it at least can't be worse than the version of steve in master.

jeffmoffitt, 2022-08-30:
This feature has now been added to the new steve in pull request 123

closed 2022-08-30 by jeffmoffitt