ZhuangLab/storm-control #63: Dave does not recognize some errors in Kilroy

leonardosepulveda, 2017-10-29:
(Add this in the future) I had errors in the Kilroy configuration file (Typo in one of the command names). Dave did not recognize this error and just stayed indefinitely in the step prior to the typo. It will be great to connect Kilroy and dave so that dave issues a warning or error message when Kilroy crashes, so I know that the run has crashed.

HazenBabcock, 2017-10-30:
Did you run the validator in Dave before you started the run?

leonardosepulveda, 2017-10-30:
I think I did, but I will verify.

leonardosepulveda, 2017-10-30:
Hi Hazen,

I verified, and the validator did not catch the error.

The error was the following: I had this code in the Kilroy config file:

<protocol name = "Hybridize 8">
<valve duration = "5">Set Hyb 8</valve>
<pump duration = "180">1 mL/min</pump>
<pump duration = "300">0.1 mL/min</pump>
<pump duration = "600">Stop Flow</pump>
<valve duration = "5">Seal</valve>
<valve duration = "5">Wash Buffer</valve>
<pump duration = "90">0.5 mL/min</pump>
<pump duration = "300">0.1 mL/min</pump>
<pump duration = "60">Stop Flow</pump>
<valve duration = "5">Seal</valve>
<valve duration = "5">Image Buffer</valve>
<pump duration = "180">0.5 mL/min</pump>
<pump duration = "60">Stop Flow</pump>
<valve duration = "5">Seal</valve>
</protocol>

The error here is that I put "1 mL/min" instead of "1.0 mL/min". So, when
the Hybridize 8 code was called, Kilroy stayed in the previous step (Set
Hyb 8) indefinitely instead of crashing. So I arrived the next morning and
my run was stuck in that step all night. Kilroy crashed, it says that  "1
mL/min" was not a valid command. It is just that dave did not got that
info.

Leonardo

jeffmoffitt, 2017-10-30:
I see the issue.  When you ask dave to validate a kilroy protocol request,
it checks only the name of the protocol. Kilroy does not have a way to
validate that all commands within a protocol are valid.

You should add it as an issue.

Best
Jeff

HazenBabcock, 2017-10-31:
This is also a related issue to #58.