ZhuangLab/storm-control #74: Hamamtsu camera raise DCAMException("DCAM initialization failed.")

CodeNeuropathUSZ, 2018-02-21:
Hi,
I am trying to use your code with a Hamamatsu Orca Flash 4.0 V2
the camera is working properly with the original HCI software, I have installed both the driver 14.* and 17.* but experiencing the same behaviour.

in particular I am trying the raise the hamamatsu_camera.py script, which is raising this exception:

Traceback (most recent call last):
  File "hamamatsu_camera.py", line 216, in <module>
    raise DCAMException("DCAM initialization failed.")
__main__.DCAMException: 'DCAM initialization failed.'

Any suggestions?

HazenBabcock, 2018-02-21:
Sorry, I can only suggest the obvious.. Is the camera turned on? Is any other software running that is using the camera?

HazenBabcock, 2018-02-21:
Can you pull master and try again? I changed it to print out the error code when initialization fails. Hopefully that will tell us something useful.

CodeNeuropathUSZ, 2018-02-21:
Hi,
the camera is on, I took care to close the HCI software to avoid that it is reading it. If I use the driver version 16 I got this:
found: 1 cameras
1.0
camera 0 model: C13440-20C
Testing fixed length acquisition
max camera backlog was: 0
Traceback (most recent call last):
  File "hamamatsu_camera.py", line 996, in <module>
    hcam.startAcquisition()
  File "hamamatsu_camera.py", line 882, in startAcquisition
    paramattach),
OSError: exception: access violation reading 0x00000010
So it is a driver issue also. I am trying now your new commit and also the driver version 15

CodeNeuropathUSZ, 2018-02-21:
With the driver version 16 and the lastest commit,  the camera is recognized but not going further than than the line
        paramattach = DCAMBUF_ATTACH(0, DCAMBUF_ATTACHKIND_FRAME,
                self.hcam_ptr, self.number_image_buffers)
        paramattach.size = ctypes.sizeof(paramattach)
Raising the OSError: exception: access violation reading 0x00000010
I will try with the driver version 15.

HazenBabcock, 2018-02-21:
I don't think it is the driver. This is a very puzzling error as it seems to be complaining that we can't set the size field of the DCAMBUF_ATTACH structure that we just created. 

@emanuega any idea what might be going on?

emanuega, 2018-02-21:
That error is especially odd considering that it must be successfully
creating structures and setting the size in the same manner in order to get
to that point in the code (such as at lines 212-213, 292-293,...).  What is
the full output for the error? What is the python version and the operating
system?

HazenBabcock, 2018-02-21:
Also, how much memory does the computer have? As written this code is going to try and use 2+ GB.

alecrimi, 2018-02-21:
python 3.6 (there were here and there some "print" which where for 2.7), is this the issue?
The machine is very powerful (Half Tera), I doubt it is memory.

HazenBabcock, 2018-02-21:
If there were print statements in `hamamatsu_camera.py` then we might be talking about quite different versions. Seems unlikely though as `paramattach.size = ctypes.sizeof(paramattach)` was added quite recently. We're assuming that you are using storm-control/master branch, latest commit fb613d40b99f213b2167305a39a97cec4a390987 with no additional modifications.

64 bit Windows? 64 bit Python?

CodeNeuropathUSZ, 2018-02-26:
Windows 64, but I just noticed : "Python 3.6.4 (v3.6.4:d48eceb, Dec 19 2017, 06:04:45) [MSC v.1900 32 bit (Intel)] on win32". Is this the cause?

CodeNeuropathUSZ, 2018-02-26:
with python 3.7.0 64 bits. I receive this error now
`found: 1 cameras
Traceback (most recent call last):
  File "hamamatsu_camera.py", line 919, in <module>
    hcam = HamamatsuCameraMR(camera_id = 0)
  File "hamamatsu_camera.py", line 802, in __init__
    super().__init__(**kwds)
  File "hamamatsu_camera.py", line 293, in __init__
    self.properties = self.getCameraProperties()
  File "hamamatsu_camera.py", line 348, in getCameraProperties
    ctypes.c_int32(DCAMPROP_OPTION_NEAREST))
ctypes.ArgumentError: argument 1: <class 'OverflowError'>: int too long to convert`

HazenBabcock, 2018-02-26:
Which 3.7.0? I just tested with Python 3.7.0b1 and it ran without issues.

$ python
Python 3.7.0b1 (v3.7.0b1:9561d7f, Jan 31 2018, 07:26:34) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.

$ python hamamatsu_camera.py
found: 1 cameras
1.0
camera 0 model: C11440-22C
Testing fixed length acquisition
max camera backlog was: 0
Frames loaded: 1
Wait time: 0.02471568899298419
0 [221 221 239 229 228]
Frames loaded: 2
Wait time: 0.0037453092277765854
1 [220 188 218 225 220]
2 [217 228 185 180 190]
..

Python 3.6.1 also works.
Python 3.6.1 (v3.6.1:69c0db5, Mar 21 2017, 18:41:36) [MSC v.1900 64 bit (AMD64)] on win32

CodeNeuropathUSZ, 2018-02-27:
We tested the script on a Windows7 machine and it works smoothly. The culprit is something in Windows10

HazenBabcock, 2018-02-27:
Interesting. Almost all of our acquisition computers are running Windows 7. I will see if I can find a Windows 10 machine to test on.

HazenBabcock, 2018-02-28:
It looks like we do have a Windows 10 machine, but it is in use until next week, so that is likely the soonest I will be able to investigate further. If that is too long to wait, and depending on your goals, you might try:

- Use the python2 version in the python2_pyqt4 branch.
- Revert to 2be3b94aa4185d3104fe6d62f29a8c547b56e9f3 as this is (more or less) the last version before we switched completely to the DCAM v4 API.
- Try the version in this project: [uutrack](http://uutrack.readthedocs.io/en/latest/index.html)

Any of these could be fine if all you want is a Python driver for your camera.

alecrimi, 2018-03-12:
Hi,
we have issues with uutrack as well, reverting to the older version did not change. Unfortunately, using python2 is out of discussion for us as we have many things in python3. I hope it is not that the only solution.

HazenBabcock, 2018-03-12:
Apologies, but I'm not going to be able to look into this for at least 2-3 more weeks. Apparently the setup with the Windows 10 machine is going to be in use continuously, and they don't have any time to spare for me to test.

I could try and help you sooner via Skype or remote desktop, if that works then e-mail me at the address on my Github account.

HazenBabcock, 2018-03-13:
Can you try the latest version? I was able to reproduce the bug that you are seeing on Windows 10 and I've hopefully fixed it now.

HazenBabcock, 2018-03-26:
Open this again if it still does not work.

closed 2018-03-26 by HazenBabcock