ZhuangLab/storm-control #79: Invalid command generated when Dave requests a parameter that has not been initialized

jeffmoffitt, 2018-05-12:
Dave generates an invalid command when validating a command for set parameters if that parameter has been loaded into hal but not yet selected. 

This issue may arise from the extra time that hal requires to initialize the parameter when it is activated for the first time.

jeffmoffitt, 2018-05-12:
As an additional aspect of this issue, Dave with crash when a second attempt is made to validate a sequence after this issue occurs.  Otherwise, Dave is capable of validating a sequence that has previously failed to validate (either due to a missing parameter or existing file).

HazenBabcock, 2018-05-13:
Could you post the trace back(s)?

jeffmoffitt, 2018-05-13:
There is no trace back.  Dave simply mistakes a valid command for an
invalid one.

Dave then freezes when validation is run again.  And, unfortunately, there
is no trace back either.

Best
Jeff

HazenBabcock, 2018-05-13:
Does it work properly if you set 'parallel_mode' in tcp_control to False?

HazenBabcock, 2018-05-13:
Ignore the above. This might be related to #84? If not I'm going to need a simple action sequence that duplicates the problem, or an in person demonstration of the problem.

closed 2018-05-22 by jeffmoffitt