ZhuangLab/storm-control #83: Hal crashes when pressing enter when in the X or Y text box of the Move To of stage control

emanuega, 2018-05-13:
Hal will crash if entered is pressed when editing the X or Y position in the Move To box of stage control. The expected behavior is for the stage to move to the specified position.

The traceback is:

Traceback (most recent call last):
  File "C:\software\storm-control\storm_control\hal4000\stage\stage.py", line 15
3, in handleAddButton
    self.ui.saveComboBox.addItem("{0:.1f}, {1:.1f}".format(x, y), [x, y])
ValueError: Unknown format code 'f' for object of type 'str'
Traceback (most recent call last):
  File "C:\software\storm-control\storm_control\sc_hardware\crystalTechnologies\
AOTF32Bit.py", line 21, in <module>
    next_cmd = my_socket.recv(1024).decode(encoding)
socket.error: [Errno 10054] An existing connection was forcibly closed by the re
mote host

HazenBabcock, 2018-05-13:
There are two problems here. I think I've fixed the first one in the handleAddButton() method. I had not updated it to handle the new format of the stage position (it is now a dictionary not a list). The other problem is that the 'default' button, the one that will 'pressed' when you press return is the add (position) button. When you click in the move to group box the 'go' button should become the default in order for it to work as you expect.

HazenBabcock, 2018-05-13:
I changed the UI so that the 'go' button is the default button. Close this if this fix works for you.

closed 2018-05-13 by emanuega