ZhuangLab/storm-control #89: How to cite you?

aquilesC, 2019-02-02:
Is it possible to cite this package in a scientific publication?

HazenBabcock, 2019-02-02:
If you are allowed to cite a web-page then I would use this:

`
@misc{SCnt:2017,
   title = {Storm-Control, microscope control software},
   url = {https://github.com/ZhuangLab/storm-control},
   year = {2019}
}
`

Otherwise there isn't really a publication specifically about this project. The closest might be:

`
@article {Babcock:2018,
   author = {Hazen P. Babcock},
   title = {Multiplane and spectrally-resolved single molecule localization microscopy with industrial grade \uppercase{CMOS} cameras},
   journal = {Scientific Reports},
   volume = {8},
   year = {2018},
}   
`

Which briefly discusses this package in the methods. If neither of these seem appropriate that's okay.

aquilesC, 2019-02-03:
Thanks! For sure a website can be mentioned. 

Maybe it would be interesting for you a platform like [Zenodo](https://zenodo.org/) that gives you a citable DOI also for packages? You can see, for example, what the guys at [Trackpy](https://zenodo.org/record/1226458) did.

closed 2019-02-03 by aquilesC

HazenBabcock, 2019-07-01:
Thank you for the suggestion, I added Zenodo badge.