ZhuangLab/storm-control #90: AttributeError: function 'compare' not found;

hollxiao, 2019-04-05:
Doctor HazenBabcock :
Hello! 

After I was runing "python hal-4000.py xml/none_config.xml",  these mistakes trouble me a lot. I try my best to fix it but failed. I wonder whether you can help me. 
Thanks a lot.

below is the bug.

···
(storm_control) C:\Users\shen\storm-control-master\storm_control\hal4000>python hal4000.py xml/none_config.xml
Logging Error! Could not open C:/Data/logs/hal4000_37.out
  Logging is disabled.
Loading modules
  hal
  aotf
  camera1
  daq
  display
Traceback (most recent call last):
  File "hal4000.py", line 756, in <module>
    parameters_file_name = args.default_xml)
  File "hal4000.py", line 447, in __init__
    a_module = importlib.import_module(module_params.get("module_name"))
  File "C:\Users\shen\Anaconda3\envs\storm_control\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\display\display.py", line 13, in <module>
    import storm_control.hal4000.display.cameraViewers as cameraViewers
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\display\cameraViewers.py", line 11, in <module>
    import storm_control.hal4000.display.cameraFrameViewer as cameraFrameViewer
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\display\cameraFrameViewer.py", line 27, in <module>
    import storm_control.hal4000.qtWidgets.qtCameraGraphicsScene as qtCameraGraphicsScene
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\qtWidgets\qtCameraGraphicsScene.py", line 13, in <module>
    import storm_control.hal4000.halLib.c_image_manipulation_c as c_image
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\halLib\c_image_manipulation_c.py", line 27, in <module>
    ctypes.c_int]
  File "C:\Users\shen\Anaconda3\envs\storm_control\lib\ctypes\__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)
  File "C:\Users\shen\Anaconda3\envs\storm_control\lib\ctypes\__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'compare' not found

···

HazenBabcock, 2019-04-05:
Do you have more than one version of storm-control installed on your computer?

This seems to be a problem with the storm-control-master/storm_control/c_libraries/c_image_manipulation.dll library. This library is supposed to have a function called 'compare' but for some reason this function is not found. So one possibility is that an older version of the library from some other version of storm-control is getting loaded instead.

If you add print(c_lib_path) to the loadCLibrary() function in storm-control-master/storm_control/c_libraries, what do you see?

```
def loadCLibrary(library_filename):

    #
    # c_lib_path is something like:
    #    /usr/lib/python3.5/site-packages/storm_control
    #
    c_lib_path = os.path.dirname(os.path.abspath(storm_control.__file__))

    # All the C libraries are in the c_libraries directory.
    c_lib_path = os.path.join(c_lib_path, "c_libraries")

    print(c_lib_path)

    # Windows.
    if (sys.platform == "win32"):
```

hollxiao, 2019-04-05:
I did as you said, The output is just shown as "C:\Users\shen\storm-control-master\storm_control\c_libraries" and the others don't change.

```
Loading modules
  hal
  aotf
  camera1
  daq
  display
C:\Users\shen\storm-control-master\storm_control\c_libraries
Traceback (most recent call last):
  File "hal4000.py", line 758, in <module>
    parameters_file_name = args.default_xml)
  File "hal4000.py", line 447, in __init__
    a_module = importlib.import_module(module_params.get("module_name"))
  File "C:\Users\shen\Anaconda3\envs\storm_control\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\display\display.py", line 13, in <module>
    import storm_control.hal4000.display.cameraViewers as cameraViewers
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\display\cameraViewers.py", line 11, in <module>
    import storm_control.hal4000.display.cameraFrameViewer as cameraFrameViewer
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\display\cameraFrameViewer.py", line 27, in <module>
    import storm_control.hal4000.qtWidgets.qtCameraGraphicsScene as qtCameraGraphicsScene
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\qtWidgets\qtCameraGraphicsScene.py", line 13, in <module>
    import storm_control.hal4000.halLib.c_image_manipulation_c as c_image
  File "C:\Users\shen\storm-control-master\storm_control\hal4000\halLib\c_image_manipulation_c.py", line 27, in <module>
    ctypes.c_int]
  File "C:\Users\shen\Anaconda3\envs\storm_control\lib\ctypes\__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)
  File "C:\Users\shen\Anaconda3\envs\storm_control\lib\ctypes\__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'compare' not found

hollxiao, 2019-04-05:
My operate system is win7 64bit. 
I look up the "c_image_manipulation_c.py"，the “compare” function dose exist in the file.
debug shows module_params.parameters['module_name'] = {NameError}name 'module_params' is not defined .
so what should I do next step for this.

def compare(image1, image2):
    """
    This does a bytewise comparison of two images.
    
    image1 - The first image.
    image2 - The second image.

    Returns the number of differences greater than 1.    
    """
    if (image1.size != image2.size):
        print("Images are not the same size!")
        return

    return image_manip.compare(image1, image2, image1.size)

HazenBabcock, 2019-04-05:
What version of Python and Anaconda are you using?

The problem I think is that the `compare` function is not found in the C library `c_image_manipulation.dll`. Did you attempt to compile `c_image_manipulation.c` yourself? Or are you using the version that is in the storm-control repository?

hollxiao, 2019-04-05:
I find out the problem is that I have the **compare** function in my **c_libraries**  but I can't use it.
I lack of the operate system knowledge so I can't fully understand the dill file and don't know how to compile it. 
And later I'll try it.

hollxiao, 2019-04-05:
It's all 64bit.
[image]
```
import sys
sys.platform
#  win32
``` 

I find out the problem is that I have the **compare** function in my **c_libraries** but I can't use it.
I lack of the operate system knowledge so I can't fully understand the dill file and don't know how to compile it.
And  I'll try it.

HazenBabcock, 2019-04-05:
Not sure what the issue as I cannot reproduce it. I did the following on a 64 bit Windows 7 computer.

1. Download and install Anaconda3-5.2.0-Windows-x86_64.exe
2. Download storm-control from github, and extract to c:\users\hazen\storm-control-master
3. Follow the install instructions [here](https://github.com/ZhuangLab/storm-control/blob/master/storm_control/hal4000/INSTALL.txt) for conda.
4. Add a storm_control.pth file to my virtual environment (in Lib\site-packages).

```
(C:\Users\Hazen\Anaconda3-5.2.0\envs\pyenv\ac3_3p6p4) C:\Users\Hazen>python
Python 3.6.4 |Anaconda, Inc.| (default, Mar 12 2018, 20:20:50) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

```
(C:\Users\Hazen\Anaconda3-5.2.0\envs\pyenv\ac3_3p6p4) C:\Users\Hazen\storm-control-master\storm_control\hal4000>python hal4000.py xml\none_config.xml
fatal: Not a git repository (or any of the parent directories): .git
Loading modules
  hal
  aotf
  camera1
  daq
  display
  feeds
  film
  filter_wheel
  focuslock
  illumination
  mosaic
  none_405
  none_em_filter_wheel
  none_irlaser
  none_qpd
  none_stage
  none_wheel1
  none_zstage
  progressions
  scmos_cal
  settings
  spotcounter
  stage
  tcp_control
  timing

core 'configure1'
camera1 'initial parameters'
daq 'wait for'
display 'add to ui'
...
```

And it all worked as expected.

hollxiao, 2019-04-06:
Thank you sir. I'll keep trying to fix it. If I find where the problem is , I’ll post here.