ZhuangLab/storm-control #91: mosaic flip/transpose not working

BogdanBintu, 2019-04-25:
In Hal4000 when editing a parameters xml, from the mosaic tab when changing the flip horizontal/vertical and transpose from False to True nothing happens. These variables are dummies.

HazenBabcock, 2019-04-25:
Not sure what you mean. The current values should be saved in the movie XML that Steve will read when it loads the movie to display. It is true that they don't do anything in HAL. If you want to change the orientation of the HAL display then you need to change the values in camera section of the hardware XML file. These cannot be changed using the HAL's editor. The idea was that you'd set them once to match the X/Y convention of the stage.

BogdanBintu, 2019-04-25:
[image]

Variables flip_horizontal, flip_vertical and transpose are dead.

HazenBabcock, 2019-04-25:
Dead? I think what you meant to say is that when you press 'update' the values reset to their initial value. Yes this is an issue and thank you for pointing it out. What I think should happen is that these parameters should not even be displayed in the editor as they are not actually editable. They should be set correctly in the hardware XML file.

HazenBabcock, 2019-04-25:
FWIW, only the 'objective' parameter is actually editable.

BogdanBintu, 2019-04-25:
Great. Sorry, I got confused what is the indented way to change the flip and transpose properties. I agree with you, it is more natural for these to be hardware settings dependent on the scope. Thanks for clarifying, problem solved.

HazenBabcock, 2019-04-25:
Okay I will fix this so these parameters don't appear in the editor. Please leave this issue open as a reminder.

HazenBabcock, 2019-04-26:
This should be fixed now, sorry for the confusion.

closed 2019-04-26 by HazenBabcock