ZhuangLab/storm-control #93: Steve positions are not compatible with Dave

BogdanBintu, 2019-05-13:
On Windows machines, upon saving the positions selected in Steve, a new line is added between positions. This causes Dave an error when trying to read the empty lines in the positions file saved by Steve.

HazenBabcock, 2019-05-13:
This used to work.. Steve is adding a "\r\n" to each line so that if you load the text file in notepad it will display correctly. Anyway at least two problem here. 

1. The [docs](https://docs.python.org/3.6/library/os.html) say that one should always use "\n" when writing text (see os.linesep). 
2. Dave should ignore blank lines. It is easy to accidentally add a trailing blank line to the list with a text editor and this will also mess things up.

My preference would to fix (1) in the 'steve_v2' branch, merge this and then fix (2). However as far as I know no one has done any testing on this branch besides me..

HazenBabcock, 2019-10-23:
This should be fixed as of 2e11e1b6d42f1e76c268de2030d8884223e2bd72.

HazenBabcock, 2019-10-25:
Reopen if this continues to be a problem.

closed 2019-10-25 by HazenBabcock