ZhuangLab/storm-control #96: Is there a module for z calibration?

Ikru488, 2020-01-09:
Hi, I was wondering if there is an existing module for z calibration ? zee_calibrator appears to be deprecated, and the z_calibrator inside storm_analysis assumes that number of z steps is equal to the no. of frames in the movie. Which means I should ideally be acquiring the astigmatism data by synchronizing camera and microscope.

HazenBabcock, 2020-01-09:
Did you see this Jupyter notebook? This is the approach that I would recommend following.
[z-calibration notebook](https://github.com/ZhuangLab/storm-analysis/blob/master/jupyter_notebooks/dao3d_zcal.ipynb)

You specify the z values for each frame in a text file, there are no assumptions about the number of z steps or how they correspond to frames in the movie.

Yes, zee_calibrator is deprecated.

Ikru488, 2020-01-12:
I tried the Jupyter notebook. Yup, this makes sense. 
It's just that I'm doing the scan without a z stage, which is why I was thinking about synchronizing.
Thanks anyway!

closed 2020-01-12 by Ikru488

HazenBabcock, 2020-01-14:
If the z position difference between frames is small (< 10nm) then I don't think synchronization is necessary. FWIW, even with a (piezo) z-stage storm-control is only software synchronizing when you do a "calibration" scan. This synchronization won't be very accurate compared to say "Hardware Z-Scan" mode where the piezo motion is actually synchronized with the camera (up to the piezo response time).