ZhuangLab/storm-control #97: Scanning for focus when original offset is close to zero

seichhorn, 2020-01-22:
I'm finding that when I lock my focus lock with an offset very close to zero, if I lose focus and enter the scan_focus routine I end up erroneously accepting the first measurement. This seems to be because 1) the uc480camera is returning total_good = 0, and 2) the difference between my IR spots returns as 0 owing to no good fits. The difference of 0 passes the threshold for the difference relative to the offset in the handleQPDUpdate step of the ScanMixIn class. Is there a better value to return here instead of 0?

jeffmoffitt, 2020-01-22:
Do you have this problem with the current version? I encountered what I think may be the same problem, and we pushed a fix recently.

seichhorn, 2020-01-22:
Thanks, I was on another branch and missed that commit, but it seems to have fixed my issue.

closed 2020-01-22 by seichhorn

HazenBabcock, 2020-01-22:
I'm still puzzled why the "sum" check is not catching this. When there are no spots on the camera the "sum" signal should be below  "minimum_sum". Does no one set "minimum_sum" properly? Or does this just now work for some other reason?

seichhorn, 2020-01-22:
My focus lock sum that is only ~2-fold lower when my spots are badly out of focus/undetected, and couldn't find a way to set it that guaranteed I didn't erroneously pass the minimum_sum threshold during the scan despite being out of focus.

HazenBabcock, 2020-01-22:
Okay, thanks.