ZhuangLab/storm-control #107: Windows fatal exception when running tests

emanuega, 2020-04-02:
When I try to run the tests on a fresh install I get a "Windows fatal exception: access violation" error at the first test in test_hal_film.py. Do you know how to resolve this issue?

I've included the output below. 

================================================= test session starts =================================================
platform win32 -- Python 3.8.2, pytest-5.4.1, py-1.8.1, pluggy-0.13.1
PyQt5 5.14.1 -- Qt runtime 5.14.1 -- Qt compiled 5.14.1
rootdir: C:\Users\GeorgeEmanuel\code\storm-control
plugins: qt-3.3.0
collected 109 items

sc_hardware\test\test_af_lock_c.py .........                                                                     [  8%]
sc_hardware\test\test_corr_2d_gauss_c.py .....                                                                   [ 12%]
sc_hardware\test\test_corr_lock_c2dg.py .                                                                        [ 13%]
test\test_buffered_1.py ..                                                                                       [ 15%]
test\test_c_libraries.py ...                                                                                     [ 18%]
test\test_corr_lock.py .                                                                                         [ 19%]
test\test_dave_starts.py .                                                                                       [ 20%]
test\test_dave_xml_generators.py .                                                                               [ 21%]
test\test_hal_error.py ..                                                                                        [ 22%]
test\test_hal_film.py Windows fatal exception: access violation

Current thread 0x00006ee0 (most recent call first):
  File "C:\Users\GeorgeEmanuel\code\storm-control\storm_control\test\hal\standardHalTest.py", line 35 in halTest
  File "C:\Users\GeorgeEmanuel\code\storm-control\storm_control\test\test_hal_film.py", line 21 in test_hal_film_1
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\python.py", line 184 in pytest_pyfunc_call
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 84 in <lambda>
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\python.py", line 1479 in runtest
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\runner.py", line 135 in pytest_runtest_call
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 84 in <lambda>
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\runner.py", line 217 in <lambda>
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\runner.py", line 244 in from_call
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\runner.py", line 216 in call_runtest_hook
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\runner.py", line 186 in call_and_report
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\runner.py", line 100 in runtestprotocol
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\runner.py", line 85 in pytest_runtest_protocol
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 84 in <lambda>
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\main.py", line 272 in pytest_runtestloop
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 84 in <lambda>
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\main.py", line 247 in _main
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\main.py", line 191 in wrap_session
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\main.py", line 240 in pytest_cmdline_main
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 84 in <lambda>
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\site-packages\_pytest\config\__init__.py", line 124 in main
  File "C:\Users\GeorgeEmanuel\Anaconda3\envs\sc_env\Scripts\pytest.exe\__main__.py", line 7 in <module>
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\runpy.py", line 86 in _run_code
  File "c:\users\georgeemanuel\anaconda3\envs\sc_env\lib\runpy.py", line 193 in _run_module_as_main

HazenBabcock, 2020-04-02:
Are you running each test in it's process, see `storm_control/test/README.txt`? This works on linux, but I've never tried it on Windows, if it is even possible.

At various times I've tried to figure out what storm-control was doing that breaks the testing infrastructure but I never made much progress. Simpler PyQt programs that use QThreads, QRunnables, etc.. seem to behave fine even when run multiple times in the same process. This is also why I was never got anywhere with continuous integration.

emanuega, 2020-04-02:
Thanks for pointing out the README for the tests. It looks like fork isn't available on windows. 

INTERNALERROR> AttributeError: module 'os' has no attribute 'fork'

HazenBabcock, 2020-04-02:
Virtualbox :). Anyway, yes the fact that the tests can only be run on linux, each in their own process is bad. Wish I knew what the problem is, but I really have no idea. All I know is that the tests fail on `app.exec_()`, which isn't really much to go on.